freshollie/UsbGps4Droid #7: UsbGPS stopped: too many connection problems

kalon33, 2017-08-24:
Using a USB GPS module based on a ublox LEA M8P, I got the error "too many connection problems". GPS is working properly with u-center on Windows 10.

kalon33, 2017-08-24:
[image]
[image]

freshollie, 2017-08-25:
This service is only designed to work with USB SERF protocol devices. Such as http://usglobalsat.com/p-688-bu-353-s4.aspx

I can't find any information about your device being a SERF device. Do you have documentation on its protocol?

freshollie, 2017-08-25:
I see this device supports NMEA. Is it possible I could get a log? I have only been testing this on http://usglobalsat.com/p-688-bu-353-s4.aspx.

Do any other apps in the playstore work with this device?

kalon33, 2017-08-25:
Yes, it supports NMEA, outputting various versions of it, currently set as 4.1 version. Do you want an NMEA output log? 

No, I couldn't find other app on the playstore working with USB devices, when this one is properly detecting the device and naming it. 

Ublox GPS chips works with three protocols: NMEA, proprietary UBX one, and another one (RTCM if I remember well). Here are protocol specs for mine, M8 series, which supports notably GPS, Glonass, Beidu and Galileo: https://www.u-blox.com/sites/default/files/products/documents/u-blox8-M8_ReceiverDescrProtSpec_(UBX-13003221)_Public.pdf 

----- Mail original -----

freshollie, 2017-08-25:
For now I'm going to concede that I will not be able to make this device work due to it having a vast array of protocols which need to be selected and I have no way of testing.

If you want this app to work I suggest buying the GPS using I linked.

Other options are you edit the driver to work with your device and then send me a pull request once completed.

kalon33, 2017-08-25:
I don't think they really need to be selected: by default, and if you don't have any manner to store parameters on the chip (flash or battery to keep RAM on when chip is not On), which is the majority of the devices I've seen, device outputs NMEA. 

That said, is that complex work to edit the driver to work with my device? If it is, I may have a look at sending a similar device to you to give you some way to test (if you want that? and have time to work on this). 

----- Mail original -----

freshollie, 2017-08-25:
I mean yeah I could probably edit the driver if I had the device, but editing a driver for a device which I don't own would be nearly impossible. 

Can you send me the android log file after you get the too many connection problems?

That error is raised when the device is not outputting any info at all. All this service does is literally read the data output directly strings, and it sounds like your device is sending none (Possibly waiting for the correct start signal)

I don't want you to send me anything expensive as I have no guarantee of this working. Though if you are still serious about sending me a device to work with then that can be arranged.

kalon33, 2017-08-25:
What kind of logfile do you need? adb logcat output? Anything else that would be useful for you? I will have a look to give you these over the weekend. 

Apparently this device is detected as a USB serial one (creates a COM port on Windows, and equivalent /dev/tty*** on Ubuntu Linux). 

If you think there's a way to get the device to work if you have it in hands, there's no problem for me to send mine or buy another one to send it to you. Supporting it would probably benefit to the community somehow so that's not wasted :) 

----- Mail original -----

freshollie, 2017-08-25:
Android has its own USB wrapper which doesn't require access to /dev/tty***. As this cannot be done without root. See https://developer.android.com/reference/android/hardware/usb/package-summary.html

The logcat would be perfect. I have several debug lines which I can read from the output and maybe understand the problem.

Can you cat /dev/tty*** on your computer and also send me the output?

kalon33, 2017-08-27:
So here is the [cat /dev/ttyACM0](https://github.com/freshollie/UsbGps4Droid/files/1254265/gps_output.txt)

adb logcat to come.

kalon33, 2017-08-27:
[adb logcat -d](https://github.com/freshollie/UsbGps4Droid/files/1254276/logcat.txt)

Is that the one you want?

freshollie, 2017-08-27:
Perfect. The cat shows that the device is outputting the correct data, the logcat shows where the error is. I will look into fixing an release a test build today!

kalon33, 2017-08-27:
Side question, does the app parses all NMEA messages? Does it support all GNSS constellation messages (from GPS, but also from GLONASS, Beidu and Galileo systems)? This device supports them all (as stated in the pdf file I sent). 

Thanks for the fix :)

freshollie, 2017-08-27:
I will get back to you on that. Please download and install [this apk](https://github.com/freshollie/UsbGps4Droid/raw/fix-endpoint-errors/app/build/outputs/apk/app-debug.apk) for do another adb logcat. Notice the error message on screen has now changed.

freshollie, 2017-08-27:
As per. [NmeaParser](https://github.com/freshollie/UsbGps4Droid/blob/master/app/src/main/java/org/broeuschmeul/android/gps/nmea/util/NmeaParser.java) the only information which is actually collected from the device is

- GPGGA
- GPRMC

These provide all the needed information to update the android GPS location.

All other information can be written to a file if specified (Enable/Disable NMEA log)

kalon33, 2017-08-27:
Indeed, message has changed:
[image]

Here is the [new logcat](https://github.com/freshollie/UsbGps4Droid/files/1254402/logcat.txt)

Thanks

freshollie, 2017-08-27:
Thanks, can you download the app from the same link and send me the logcat again :)

kalon33, 2017-08-27:
[Here](https://github.com/freshollie/UsbGps4Droid/files/1254426/logcat.txt) it is :)

freshollie, 2017-08-27:
Same again please, what I am trying to do is work out what interface the device requires.

kalon33, 2017-08-27:
[Here](https://github.com/freshollie/UsbGps4Droid/files/1254482/logcat.txt) it is again :)

freshollie, 2017-08-27:
Hey that was working! Leave it going!

freshollie, 2017-08-27:
The log shows that you disconnected the device.

kalon33, 2017-08-27:
Android disconnected itself, asking me again if I authorize usbgps to use it. I launched it again after the logcat, left it running. I have NMEA data in the log, but N/A for everything: number of satellites, accuracy, location, altitude and Gps Time.

kalon33, 2017-08-27:
Other android apps that are normally giving satellites list with internal GPS doesn't provide any info now.

freshollie, 2017-08-27:
Reading the log, it looks like the NMEA protocol being read back has not been implemented into the driver.

> Other android apps that are normally giving satellites list with internal GPS doesn't provide any info now.

I don't understand what you mean this.

freshollie, 2017-08-27:
I've built an APK at the same link to work for added NMEA set, try that. Leave the GPS somewhere where it can get a fix. The log showed now GPS fix.

kalon33, 2017-08-27:
In some other android GPS apps (for internal GPS devices), you can get a list of current satellites used, and various info about GPS. For now I don't have them with the external one. Is that expected?

Now I have proper position fix, with altitude and number of satellites used in UsbGPS (the latter one not outside UsbGPS)

freshollie, 2017-08-27:
Yes, that is expected due to the driver not interpreting the list of satellites from the device, only the number.

So the device is now working perfectly?! :))

freshollie, 2017-08-27:
If you want an interpretation of the satellite positions, please make a separate issue. It will take much longer for me to produce, though it is a long term goal.

kalon33, 2017-08-27:
yep, that's now working :)

Is that expected that GPS time is a long integer, not usual time format? I'll create this issue, and, if you mind, another one to have some basic options like the ones available for sirf devices :)

freshollie, 2017-08-27:
"Gps Time" should be the same length as "Sys Time" as can be seen from the screenshots. That's great this is now working, I am surprised.

closed 2017-08-27 by freshollie

kalon33, 2017-08-27:
so this is OK, both strings look like the same kind of thing. What's the difference between altitude (MSL) and altitude? I get both from my internal chip and only MSL one from external device (but not labeled as such).

kalon33, 2017-08-27:
I don't have fix type info neither. Is that expected?

kalon33, 2017-08-27:
neither VDOP/HDOP/PDOP...

freshollie, 2017-08-27:
Do the altitude numbers differ? I just read what comes from the nmea and add it to a location object.

https://developer.android.com/reference/android/location/Location.html

Notice this object does not have the ability to set satellites or any of that info. I am pretty sure this is a limitation of android.

commit b37e0b0e (2017-08-27): Fixes and closes #7, fixed endpoint finder, fixed NMEA parser to work with all type of NMEA