freshollie/UsbGps4Droid #10: Like for SIRF options, add some UBX options to configure devices

kalon33, 2017-08-27:
As it already exists for SIRF based devices, please add some options to basically configure UBX based devices (like SBAS enable/disable, GPS/GLONASS/Beidu/Galileo enable/disable, etc...).

Thanks :)

freshollie, 2017-08-31:
Unfortunately this would be nearly impossible to write and test without a UBX based device, and I don't really have the budget to be buy a UBX device. 

I will leave this open in case something changes in the future, but for now this is not a priority, sorry!

larionovavi-stack, 2026-08-19:
This is implemented in a maintained fork: https://github.com/larionovavi-stack/UsbGps4Droid (disclosure: I maintain it).

The settings screen can now configure a u-blox receiver over UBX — measurement/navigation rate, dynamic platform model, SBAS on/off, and save-to-flash or reset-to-defaults — alongside a full UBX parser (NAV-PVT, NAV-SOL, NAV-STATUS, NAV-SVINFO) and Dead Reckoning support for the LEA-6R. See #21 for the details.

Offered upstream as #46. APK: https://github.com/larionovavi-stack/UsbGps4Droid/releases/latest