freshollie/UsbGps4Droid #8: Parse and transmit to the system more infos about satellites

kalon33, 2017-08-27:
Currently, when using an external device, we can see this kind of information in other GPS apps:
[image]
[image]

This is what would be great to be displayed, as it is from the internal phone GPS device:
[image]
[image]

Is this possible to have this kind of info? As previously, I'm ready to test and provide logs if needed :)

freshollie, 2017-08-27:
No need for you to test. If I create this it will work with any NMEA data including my own GPS unit. Incidentally what is this app called? Is it open source because creating UI is the worst part, the data is easy to gather.

kalon33, 2017-08-27:
https://github.com/barbeau/gpstest

freshollie, 2017-08-27:
Wait, I just read the title and realised that I just need to send this info to the system. I will look into this.

freshollie, 2017-08-27:
The only option would be for me to implement this into my app directly instead of an external app. I will be able to do this by potentially copying the code from this other app. I cannot send any of this data to android without root.

freshollie, 2017-08-31:
I have started work on this

sobrinho, 2018-11-20:
@freshollie have you ever finished this? :)

freshollie, 2018-11-21:
@sobrinho Yeah it's mostly completed in its branch, however it has some problems which are mostly to do with the satellite position view. If you can build the APK it should run: https://github.com/freshollie/UsbGps4Droid/tree/satellite-analysis-development