freshollie/UsbGps4Droid #12: App crashes on screen rotation on Choose Usb Gps page

neupanedipen, 2018-05-19:
#### Expected behavior<br>
The app must be stable during screen rotation on &quot;Choose Usb Gps&quot; page.<br>
#### Actual behavior<br>
But the app crashes on rotating the screen on &quot;Choose Usb Gps&quot; page.<br>
#### How to reproduce<br>
1. Install the app from releases page.<br>
2. Launch the app.<br>
3. Click on settings icon from top right corner of the app and click on &quot;Choose Usb Gps&quot; option.<br>
4. Now change the orientation of the screen to landscape mode.<br>
The app crashes.<br>
* App Version: beta-v2.1.5<br>
* Device: Samsung Galaxy J5<br>
* Operating System: Android 6.0.1<br>
#### Recording Of The Bug<br>
https://youtu.be/hfTSZ-87SCo

tobias-g1, 2018-05-20:
Replicated the issue on Android 7.0 

```
05-20 11:01:11.441 16103-16103/org.broeuschmeul.android.gps.usb.provider E/AndroidRuntime: FATAL EXCEPTION: main
    Process: org.broeuschmeul.android.gps.usb.provider, PID: 16103
    java.lang.RuntimeException: Unable to start activity ComponentInfo{org.broeuschmeul.android.gps.usb.provider/org.broeuschmeul.android.gps.usb.provider.ui.SettingsActivity}: java.lang.IllegalStateException: ListPreference requires an entries array and an entryValues array.
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2646)
        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2707)
        at android.app.ActivityThread.handleRelaunchActivity(ActivityThread.java:4483)
        at android.app.ActivityThread.-wrap19(ActivityThread.java)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1466)
        at android.os.Handler.dispatchMessage(Handler.java:102)
        at android.os.Looper.loop(Looper.java:154)
        at android.app.ActivityThread.main(ActivityThread.java:6077)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:866)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:756)
     Caused by: java.lang.IllegalStateException: ListPreference requires an entries array and an entryValues array.
        at android.preference.ListPreference.onPrepareDialogBuilder(ListPreference.java:249)
        at android.preference.DialogPreference.showDialog(DialogPreference.java:306)
        at android.preference.DialogPreference.onRestoreInstanceState(DialogPreference.java:453)
        at android.preference.ListPreference.onRestoreInstanceState(ListPreference.java:315)
        at android.preference.Preference.dispatchRestoreInstanceState(Preference.java:1819)
        at android.preference.PreferenceGroup.dispatchRestoreInstanceState(PreferenceGroup.java:325)
        at android.preference.Preference.restoreHierarchyState(Preference.java:1801)
        at android.preference.PreferenceFragment.onActivityCreated(PreferenceFragment.java:218)
        at android.app.Fragment.performActivityCreated(Fragment.java:2362)
        at android.app.FragmentManagerImpl.moveToState(FragmentManager.java:1014)
        at android.app.FragmentManagerImpl.moveToState(FragmentManager.java:1171)
        at android.app.FragmentManagerImpl.moveToState(FragmentManager.java:1153)
        at android.app.FragmentManagerImpl.dispatchActivityCreated(FragmentManager.java:2039)
        at android.app.FragmentController.dispatchActivityCreated(FragmentController.java:176)
        at android.app.Activity.performCreateCommon(Activity.java:6656)
        at android.app.Activity.performCreate(Activity.java:6664)
        at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1118)
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2599)
        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2707) 
        at android.app.ActivityThread.handleRelaunchActivity(ActivityThread.java:4483) 
        at android.app.ActivityThread.-wrap19(ActivityThread.java) 
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1466) 
        at android.os.Handler.dispatchMessage(Handler.java:102) 
        at android.os.Looper.loop(Looper.java:154) 
        at android.app.ActivityThread.main(ActivityThread.java:6077) 
        at java.lang.reflect.Method.invoke(Native Method) 
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:866) 
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:756) 
```

freshollie, 2018-05-20:
Interesting find! I had trouble replicating this, as it doesn't show up on the debug version of the APK. Thanks for the stacktrace, will look into fixing.

commit d5b89c7f (2018-05-20): Fix crashes when rotating screen with usb selection box open in settings, closes #12. Fix errors when parsing NMEA strings with bad time values
closed 2018-05-20 by freshollie