freshollie/UsbGps4Droid #13: Need help in connecting USB GPS to Android based raspberry pi 3 Model b+

RojitJohncy, 2018-11-12:
Hi,

Really need you help in connecting the external USB GPS to my Raspberry pi 3 model B+, which has android running in it. Tried several apps from the internet, but still no luck. Do you have the .apk file by any chance? as i'm very new to Android, i'm not sure about the steps of compiling the source code and generate the .apk file. Thanks in advance.

**GPS Model:** HW-658 GSM/GPRS Module USB GPS Module For Raspberry Pi.

closed 2018-11-13 by freshollie

RojitJohncy, 2018-11-13:
Thanks for your quick help :)

I have one more question > does the latest version support HW-658 GSM/GPRS Module USB GPS Module For Raspberry Pi? Bought the device from below link. the device works fine with Rasbian stretch. But not sure how to make the gps work with Android running on RPI 3 model B+.
https://www.aliexpress.com/item/HW-658-GSM-GPRS-Mould-USB-GPS-Module-For-Raspberry-Pi-A-B-A-B-Zero/32913026283.html

Thanks in advance!!

freshollie, 2018-11-13:
I couldn't possibly know. From reading the specsheet, it looks like the device uses NMEA for communication, which is what this app is designed to read. https://github.com/freshollie/UsbGps4Droid/blob/master/app/src/main/java/org/broeuschmeul/android/gps/nmea/util/NmeaParser.java

RojitJohncy, 2018-11-17:
For some reason, the USB gps which I have mentioned above is not working. Will you be able to help me?

freshollie, 2018-11-17:
Yes, what appears to be the issue? Have you got any screenshots?

RojitJohncy, 2018-11-24:
Yes, I do have pics, how do I attach it here?

freshollie, 2018-11-24:
https://blog.github.com/2015-09-25-attach-files-to-comments/

RojitJohncy, 2018-11-24:
[image]
[image]

Attaching screenshots, i just see NA for all and when i try to use maps (tried WAZE) i get error no GPS found. i doubt if the external gps is not detected by the device. i dont see the led on gps unit blinking. Is there any config changes required so that device understands an external GPS is connected? Apologies for the basic questions asked, cause i have very basic knowledge in Android development and Raspberry pi.

Thanks in advance !! :)

freshollie, 2018-11-24:
The data in the log box shows that the GPS unit is sending data to the app correctly, however the data shows no GPS fix. Your second screenshot shows that you have the service switched off.

Ensure that your GPS unit is in a position in which it can receive a GPS signal. You will see N/A in the values until a GPS fix is acquired.

RojitJohncy, 2018-11-24:
I tried enabling the service for over 1 hour, but still don't see the GPS fix. I just see values as "NA". Is there anything we can to fix gps?

freshollie, 2018-11-24:
Like I said, you must ensure that the GPS device is in an area where it can receive a signal. Have you tried it outside?

RojitJohncy, 2018-11-24:
Yes, have tried it in an open area, the same gps works with raspbian os running in it. When I switch back to Android I get the problem. ☹️

freshollie, 2018-11-24:
The screenshots which you have sent me do not show a GPS fix. Try to achieve a fix on raspbain and and then show me the same screenshot on android.

RojitJohncy, 2018-11-24:
Will send you the screenshots from raspbian os running in it. Is there a way we can confirm if the external GPS is detected on the device?

freshollie, 2018-11-24:
I don't want to see it running on Raspbian. I am trying to help show you how you might solve this problem. Make sure you get a fix in Raspbian, and then switch to android and see what the log output is from the app.

RojitJohncy, 2018-11-27:
Still no luck. :( Should we add some drivers, so that the device can detect the external GPS?