freshollie/UsbGps4Droid #14: USB permission

sobrinho, 2018-11-28:
Hi there,

Is [this](https://stackoverflow.com/questions/13726583/bypass-android-usb-host-permission-confirmation-dialog/30563253#30563253) still the way to go?

My device isn't rooted and this USB popup is really annoying.

I know that `/dev/ttyACM0` is available to be read by any user/app, that helps somehow?

freshollie, 2018-11-28:
Yeah, unfortunately it's an android issue.

closed 2018-11-28 by freshollie

sobrinho, 2018-11-28:
I'm seeing some comments on StackOverflow that you may specify the USB device properties somehow and avoid that warning.

Have you seen this before?

freshollie, 2018-11-28:
That is to do with usb accessory options. I attempted this route but it still didn't work. Why are you not able to root the device?

sobrinho, 2018-11-28:
I'm using a Tinker Board S and the options are limited, some people were able to root but the documentation is still missing across the web.

The few tutorials you will find doesn't work at all 😞