freshollie/UsbGps4Droid #16: v2.1.6 crashes at start but not v2.1.3 on LineageOS 9 possibly due to an issue with notifications

KamalW, 2019-08-23:
Thank you for the great app.

Here is the logcat output which indicates the problem:

08-23 16:08:35.229   285   502 D audio_hw_primary: out_set_parameters: enter: usecase(4: audio-ull-playback) kvpairs: routing=2
08-23 16:08:35.233   285  4026 D audio_hw_primary: start_output_stream: enter: stream(0xaa057000)usecase(4: audio-ull-playback) devices(0x2)
08-23 16:08:35.233   285  4026 D audio_hw_primary: start_output_stream: enter: usecase(4: audio-ull-playback) devices(0x2)
08-23 16:08:35.233   285  4026 D audio_hw_primary: select_devices: out_snd_device(2: speaker) in_snd_device(0: )
08-23 16:08:35.233   285  4026 W msm8974_platform: Codec backend bitwidth 16, samplerate 48000
08-23 16:08:35.233   285  4026 D hardware_info: hw_info_append_hw_type : device_name = speaker
08-23 16:08:35.233   285  4026 I msm8974_platform: platform_send_audio_calibration: sending audio calibration for snd_device(2) acdb_id(15)
08-23 16:08:35.233   285  4026 D ACDB-LOADER: ACDB -> send_audio_cal, acdb_id = 15, path =  0
08-23 16:08:35.233   285  4026 D ACDB-LOADER: ACDB -> send_adm_topology
08-23 16:08:35.233   285  4026 D ACDB-LOADER: ACDB -> ACDB_CMD_GET_AUDPROC_COMMON_TOPOLOGY_ID
08-23 16:08:35.233   285  4026 D ACDB-LOADER: ACDB -> send_asm_topology
08-23 16:08:35.233   285  4026 D ACDB-LOADER: ACDB -> ACDB_CMD_GET_AUDPROC_STREAM_TOPOLOGY_ID
08-23 16:08:35.233   285  4026 D ACDB-LOADER: ACDB -> send_audtable
08-23 16:08:35.233   285  4026 D ACDB-LOADER: ACDB -> ACDB_CMD_GET_AUDPROC_COMMON_TABLE
08-23 16:08:35.234   285  4026 D ACDB-LOADER: ACDB -> AUDIO_SET_AUDPROC_CAL
08-23 16:08:35.234   285  4026 D ACDB-LOADER: ACDB -> send_audvoltable
08-23 16:08:35.234   285  4026 D ACDB-LOADER: ACDB -> ACDB_CMD_GET_AUDPROC_GAIN_DEP_STEP_TABLE
08-23 16:08:35.234   285  4026 D         : Failed to fetch the lookup information of the device 0000000F 
08-23 16:08:35.234   285  4026 E ACDB-LOADER: Error: ACDB AudProc vol returned = -19
08-23 16:08:35.234   285  4026 D ACDB-LOADER: ACDB -> AUDIO_SET_AUDPROC_VOL_CAL
08-23 16:08:35.234   285  4026 D ACDB-LOADER: ACDB -> send_afe_cal
08-23 16:08:35.234   285  4026 D ACDB-LOADER: ACDB -> ACDB_CMD_GET_AFE_COMMON_TABLE
08-23 16:08:35.234   285  4026 D         : Failed to fetch the lookup information of the device 0000000F 
08-23 16:08:35.234   285  4026 E ACDB-LOADER: Error: ACDB AFE returned = -19
08-23 16:08:35.234   285  4026 D ACDB-LOADER: ACDB -> AUDIO_SET_AFE_CAL
08-23 16:08:35.234   285  4026 D audio_route: Apply path: speaker
08-23 16:08:35.259   285  4026 D audio_route: Apply path: audio-ull-playback
08-23 16:08:35.261   285  4026 D audio_hw_primary: select_devices: done
08-23 16:08:35.286  9268  9268 D USBGpsManager: Checking all connected devices
08-23 16:08:35.288  9268  9268 D USBGpsManager: Checking device: 13097 3725
08-23 16:08:35.288  9268  9268 D USBGpsManager: Found correct device
08-23 16:08:35.290  9268  9268 D USBGpsManager: USB GPS manager enabled
08-23 16:08:35.290  9268  9268 D USBGpsManager: starting connection and reading thread
08-23 16:08:35.290  9268  9268 D USBGpsManager: starting connection to socket task
08-23 16:08:35.294  9268  9268 D USBGpsManager: enabling mock locations provider: gps
08-23 16:08:35.313  9268  9268 W Notification: Use of stream types is deprecated for operations other than volume control
08-23 16:08:35.313  9268  9268 W Notification: See the documentation of setSound() for what to use instead with android.media.AudioAttributes to qualify your playback use case
08-23 16:08:35.320   669   696 W ActivityManager: Error showing notification for service
08-23 16:08:35.320   669   696 W ActivityManager: java.lang.RuntimeException: invalid channel for service notification: Notification(channel=null pri=0 contentView=null vibrate=null sound=null defaults=0x0 flags=0x50 color=0x00000000 vis=PRIVATE)
08-23 16:08:35.320   669   696 W ActivityManager: 	at com.android.server.am.ServiceRecord$1.run(ServiceRecord.java:663)
08-23 16:08:35.320   669   696 W ActivityManager: 	at android.os.Handler.handleCallback(Handler.java:873)
08-23 16:08:35.320   669   696 W ActivityManager: 	at android.os.Handler.dispatchMessage(Handler.java:99)
08-23 16:08:35.320   669   696 W ActivityManager: 	at android.os.Looper.loop(Looper.java:193)
08-23 16:08:35.320   669   696 W ActivityManager: 	at android.os.HandlerThread.run(HandlerThread.java:65)
08-23 16:08:35.320   669   696 W ActivityManager: 	at com.android.server.ServiceThread.run(ServiceThread.java:44)
08-23 16:08:35.336  9268  9268 D AndroidRuntime: Shutting down VM
08-23 16:08:35.337  9268  9268 E AndroidRuntime: FATAL EXCEPTION: main
08-23 16:08:35.337  9268  9268 E AndroidRuntime: Process: org.broeuschmeul.android.gps.usb.provider, PID: 9268
08-23 16:08:35.337  9268  9268 E AndroidRuntime: android.app.RemoteServiceException: Bad notification for startForeground: java.lang.RuntimeException: invalid channel for service notification: Notification(channel=null pri=0 contentView=null vibrate=null sound=null defaults=0x0 flags=0x50 color=0x00000000 vis=PRIVATE)
08-23 16:08:35.337  9268  9268 E AndroidRuntime: 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1745)
08-23 16:08:35.337  9268  9268 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:106)
08-23 16:08:35.337  9268  9268 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:193)
08-23 16:08:35.337  9268  9268 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:6718)
08-23 16:08:35.337  9268  9268 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)
08-23 16:08:35.337  9268  9268 E AndroidRuntime: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
08-23 16:08:35.337  9268  9268 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:858)
08-23 16:08:35.343   669 16264 W ActivityManager:   Force finishing activity org.broeuschmeul.android.gps.usb.provider/.ui.GpsInfoActivity
08-23 16:08:35.376  9268  9268 I Process : Sending signal. PID: 9268 SIG: 9
08-23 16:08:35.404   669 17182 I WindowManager: WIN DEATH: Window{658d874 u0 org.broeuschmeul.android.gps.usb.provider/org.broeuschmeul.android.gps.usb.provider.ui.GpsInfoActivity}
08-23 16:08:35.404   669  8009 I ActivityManager: Process org.broeuschmeul.android.gps.usb.provider (pid 9268) has died: fore TOP 
08-23 16:08:35.405   669  8009 W ActivityManager: Scheduling restart of crashed service org.broeuschmeul.android.gps.usb.provider/.driver.USBGpsProviderService in 1000ms
08-23 16:08:35.406   669   699 W libprocessgroup: kill(-9268, 9) failed: No such process
08-23 16:08:35.425   240   240 I Zygote  : Process 9268 exited due to signal (9)
08-23 16:08:35.429   306   443 W SurfaceFlinger: Attempting to set client state on removed layer: org.broeuschmeul.android.gps.usb.provider/org.broeuschmeul.android.gps.usb.provider.ui.GpsInfoActivity#0
08-23 16:08:35.429   306   443 W SurfaceFlinger: Attempting to destroy on removed layer: org.broeuschmeul.android.gps.usb.provider/org.broeuschmeul.android.gps.usb.provider.ui.GpsInfoActivity#0
08-23 16:08:35.449   669   699 W libprocessgroup: kill(-9268, 9) failed: No such process
08-23 16:08:35.449   669   699 I libprocessgroup: Successfully killed process cgroup uid 10117 pid 9268 in 43ms
08-23 16:08:35.534   669   720 W ActivityManager: setHasOverlayUi called on unknown pid: 9268
08-23 16:08:35.537   669  2362 W ActivityManager: Unable to start service Intent { act=com.android.launcher3.WINDOW_OVERLAY dat=app://com.android.launcher3:10035?v=7&cv=9 pkg=com.google.android.googlequicksearchbox } U=0: not found
08-23 16:08:35.544   669   697 W Looper  : Slow dispatch took 137ms android.ui h=com.android.server.am.ActivityManagerService$UiHandler c=null m=53
08-23 16:08:35.571  3191  3400 W Adreno-EGL: <qeglDrvAPI_eglGetConfigAttrib:607>: EGL_BAD_ATTRIBUTE
08-23 16:08:35.572  3191  3400 D vndksupport: Loading /vendor/lib/hw/gralloc.msm8974.so from current namespace instead of sphal namespace.
08-23 16:08:38.410   285   502 D audio_hw_primary: out_standby: enter: stream (0xaa057000) usecase(4: audio-ull-playback)
08-23 16:08:38.484   285   502 D hardware_info: hw_info_append_hw_type : device_name = speaker

freshollie, 2019-08-23:
Thanks for the report. I see the issue. Working on a fix.

freshollie, 2019-08-23:
Pushed a fix: eb8938ef105ee781791d23ad361c8df5287ed08f

Can you try this and let me know if it's fixed?

[UsbGps4Droid-v2.1.6-beta-debug.apk.zip](https://github.com/freshollie/UsbGps4Droid/files/3534674/UsbGps4Droid-v2.1.6-beta-debug.apk.zip)

KamalW, 2019-08-23:
It is working. Thank you.

One minor issue I see in this is that the Gps Time and Sys Time text overlaps with the Altitude text on the Overview screen. It happens when the phone is held vertically. No overlapping when the phone is held horizontally since there is enough width.

freshollie, 2019-08-27:
Ok, great! Thanks for pointing out the other bug, I will look into it. For now, I will push a new release ASAP with the fix.

freshollie, 2019-08-27:
Fixed by d44311a0145e54c2f42360e7d41dbafd1ea8c4d4

closed 2019-08-27 by freshollie