freshollie/UsbGps4Droid #18: APK of v2.2.0 lacks v1 signature/certificates

IzzySoft, 2019-08-28:
There are no certificates inside the APK, so I assume you forgot at least v1 signing. Could you please sign the APK and re-attach? Thanks!

freshollie, 2019-08-28:
Oops. Didn't really know if the v1 signatures were required anymore so I didn't tick the box. Rebuilding now.

closed 2019-08-28 by freshollie

IzzySoft, 2019-08-28:
Thanks! Explanation:

* v1 is still needed for Android 6 and below, which doesn't know about v2 as that was introduced with Android 7. So if you want to support users on devices with Android versions below 7, you ***must*** provide v1 signature.
* Starting with Android 7, v2 was introduced. Here v1 is optional and used as fallback.
* to be on the safe side, just tick both check boxes in Android Studio :wink: