freshollie/UsbGps4Droid #19: Auto Start collecting GPS data on boot up

Traxtar928, 2019-09-03:
When android boots up, auto-start the app and begin collecting GPS data for seamless start of maps application, or other applications that require GPS data to function properly.

Currently, I have to manually go into the app and start the GPS collection every boot up. 

Implementing this feature will be very useful for those that use an android device as a GPS/navigation system and the device constantly powers on/off with a vehicle/boat/etc.

freshollie, 2019-09-03:
Hi.

From the README:

> The application's background service can be set to start when the device turns on. 
> Currently the service does not automatically start when the GPS device is plugged 
> into the Android device due to more unwanted usb popups.
> 
> For now the background service can be manually started with a start service intent.
> 
> ```java
> Intent intent = new Intent();
> intent.setComponent(
> 	new Component(
> 		"org.broeuschmeul.android.gps.usb.provider",
> 		"org.broeuschmeul.android.gps.usb.provider.driver.USBGpsProviderService"
> 	)
> )
> intent.setAction("org.broeuschmeul.android.gps.usb.provider.action.START_GPS_PROVIDER")
> ```
> 
> Or via a shell command as root.
> 
> ```bash
> am startservice -a org.broeuschmeul.android.gps.usb.provider.action.START_GPS_PROVIDER -n org.broeuschmeul.android.gps.usb.provider/.driver.USBGpsProviderService
> ```
> 
> The background service will automatically close itself when the USB device is disconnected for too long.

Traxtar928, 2019-09-03:
I have 'start service on boot' selected, but I still have to manually turn on the 'Start/Stop GPS' toggle every time the android box powers on. Will the shell command you referenced from the readme address this?

freshollie, 2019-09-03:
The start service on boot should do what you desire. If it's not working, I guess it might be because the USB device is not ready? The shell command will work, but I would imagine you just need to use a tasker delay in order to make it work.

Traxtar928, 2019-09-03:
Great. I'll try it and report back. So far, 'start service on boot' is not working. If it works fine for you, using the GlobalSat BU-353-S4, maybe I should switch devices. The BeeLink GT Kind android box I'm using has a pretty quick startup time, and the ND-105C dongle stays plugged in with a start time of ~35 seconds max (averages about 15 seconds).

I've also tried it on my Samsung S9, just to ensure that it's not the device causing the startup issue, and I have to manually toggle Start/Stop GPS on that device as well, even with the 'start service on boot' checkbox enabled.

freshollie, 2019-09-03:
Btw. For your screen brightness control I recommend [autobright](https://play.google.com/store/apps/details?id=com.autobright.kevinforeman.autobright&hl=en) and headunit-controller will start this on startup too. 

Yeah, not sure exactly if the start service on boot does work for my setup, I made it a long time ago as a feature request and I didn't need to use it on my setup. Does your box do a full system start when you start your car? I might be able to add a delay to the boot start which might fix the issue.

Traxtar928, 2019-09-03:
The box does do a full system start... I have it wired to accessories, so the power is cut when the vehicle is off.

If it helps, right after it boots into android I am able to IMMEDIATELY go into the app and toggle 'Start/Stop GPS' and it will start getting satellite data right away, so it doesn't seem to be a delay issue or timing issue. Whatever happens in the software when that 'Start/stop GPS' is toggled just needs to have an option to be turned on immediately when the app starts. 

I can then automate the UsbGps4Droid app to turn on after the OS startup process in your headunit-contoller app.

In my mind, should be an easy implement... not sure about in practice, though. Great job, either way.

Traxtar928, 2019-09-05:
Did you happen to confirm that the app can auto starts collecting GPS data on boot? I want to confirm it's not my device that's causing the problems.

I also tried headunit-controller, but it caused my device to lag terribly. I posted the issue on that app's github page.

freshollie, 2019-10-08:
Hi @Traxtar928. I have not been able to confirm this. Circumstance currently means that the car I developed this system with is not used by me anymore, and I do not have access to it. Did you manage to solve these issues?

Traxtar928, 2019-11-13:
Oliver,

I have not been able to resolve it... really discouraging. I am rooted,
with TWRP, but I can't 'grant SU permission' to the app. In Xplore, I have
even moved the apk to /system/app/ and enabled all r/w/e permissions and
the owner/group is root. Still nothing.

I am honestly willing to pay you for your time to get this resolved. End
goal is to have GPS auto enabled and communicating with satellites on
startup, without any requirements for me to perform special tasks or enable
buttons.

Let me know your thoughts.

Kind regards,

Matt

Traxtar928, 2019-11-13:
The shell command in tasker returns a 255 error. I read somewhere that
android pie won't let you invoke startservice, but I'm not sure if that's
true or not.

-Matt

freshollie, 2019-11-13:
Ok. Firstly can you try this? `adb shell am start-foreground-service com.some.package.name/.YourServiceSubClassName` Instead of what I recommended.

Also, can you look into downgrading the app? https://github.com/freshollie/UsbGps4Droid/releases/tag/v2.1.3

I think this might be because on API 26+ you can't start a service in the background on boot, and v2.1.3 is on API 25.

Traxtar928, 2019-11-14:
Well, if tasker is the one executing the shell command, will that matter?

I'll report back once I've tried it. Thanks very much for the continued
assistance.

JanRSmit, 2019-12-03:
I am curious to know if there is there any result?

freshollie, 2019-12-03:
@JanRSmit yeah, this is definitely a bug since upgrading the SDK to API 26. I will get some time over christmas (hopefully) to test fixing the latest release to work with restarts.

Traxtar928, 2019-12-03:
I simply haven't had the time to implement the most recent suggestions of
version rollback and new string command.

I will report results when I do that.

invisible789, 2020-09-29:
Auto start not working for me too, at reboot it always ask for default permissions for attached usb and service is off, indeed few settings for the app like root permission for system time from gps is lost, else settings remain intact but I've to manually start the service as well
I'm scratching my head why it always ask for usb default settings on every boot
Please help

freshollie, 2020-09-29:
@invisible789 the permission popup is an android issue, it requires authorising USB devices for every reboot, or if the device disconnects. Your only solution is this: https://github.com/freshollie/UsbGps4Droid#usb-permissions-popup

As far as the service not auto starting, I never got a chance to look into the Android 9 issues. Are you running android 9?

invisible789, 2020-09-29:
No, I'm at android 7.1.2 ATV OS, I'm not that learned to Create a new SYSYTEMUI.APK, can I get such help to send you APK and you modify it and share it with so i can replace,, i may be asking too much 🥇

invisible789, 2020-09-29:
Just for info, such permission is not required in Another app , i'm not quoting its name here, untill you give permission to, its highly unethical , should i share that app name, that app doesn't require any usb permissions again & Again at every reboot , that app source code may be of help to you?

freshollie, 2020-09-29:
Yeah, would be helpful for me to have a look, if the app is open source.

invisible789, 2020-09-29:
here's play store link for the app, The app supported other than ppm until **version 2.6.0.2** and it never asked usb gps permission except once at first install of the app, it **never asks back for usb permissions at all and autostart right away**
https://play.google.com/store/apps/details?id=de.pilablu.gnsscommander&hl=en
it may be of help to community i hope :)

invisible789, 2020-09-29:
here's direct link to version 2.6.0.2
https://mega.nz/file/Kk4iXZYR#1lxtP_wRH907hD498sZ6CpkDuGITrSw7S56-qehwLcw

Traxtar928, 2020-09-29:
After installing GNSS Commander app, it runs and I see my location on the
application's built in map, with connected satellite data. It even follows
accurately while I drive.

However, my navigation apps don't see my position... I also cannot set the
mock location app to GNSS Commander... It says UsbGps (which is the app
I've been using, and I've turned off to try this new app). I cannot get
GNSS to push my mock location to my android box. The 'set GPS mock
location' checkbox is enabled in the app.

I still haven't successfully gotten auto start to work on my vehicle
android box. Every time I want to start navigation, I need to manually go
into the UsbGps so and start the service.

What am I doing wrong?

invisible789, 2020-09-30:
@Traxtar928 
Remove usbgps for to work everything properly
Than your mock location will be set to this app and auto start will work, that's what happens to me

freshollie, 2020-09-30:
I'm a bit confused why you need this app if the other one works @invisible789?

Traxtar928, 2020-09-30:
Indeed. Great question, Oliver. This other app clearly has lots of work
invested. If it auto starts and does the same thing UsbGps does, why do you
need it to work?

invisible789, 2020-09-30:
> I'm a bit confused why you need this app if the other one works @invisible789?

Actually your app uses minimum resources on the system , and it's the best part of it, moreover it's accurate & up-to-date, dear apology if i hurt you by sharing other app, looking forward for your super worked app future updates :)

Traxtar928, 2020-09-30:
That's a valid reason. I also prefer UsbGps because it's simple, to the
point, programed well, users almost no resources (I have not compared...
Have you?), and serves its purpose with no additional frills. If the auto
start worked, it would be the perfect app for my needs.

invisible789, 2020-09-30:
> That's a valid reason. I also prefer UsbGps because it's simple, to the point, programed well, users almost no resources (I have not compared... Have you?), and serves its purpose with no additional frills. If the auto start worked, it would be the perfect app for my needs.
> […](#)

Exactly ☺️

slavka93, 2020-12-15:
Hello. Experienced people helped me to launch your application in the background. But now for some reason the check mark of time synchronization with GPS is removed, maybe this is how to solve it?

#!/system/bin/sh

am start-foreground-service -a org.broeuschmeul.android.gps.usb.provider.action.START_GPS_PROVIDER -n org.broeuschmeul.android.gps.usb.provider/.driver.USBGpsProviderService &
sleep 2
am startservice -a org.broeuschmeul.android.gps.usb.provider.action.START_GPS_PROVIDER -n org.broeuschmeul.android.gps.usb.provider/.driver.USBGpsProviderService &

exit 0