freshollie/UsbGps4Droid #21: UBX support

Takumiboo, 2020-04-28:
Please implement UBX Binary support.
Like [kamabokoz's fork](https://github.com/kamabokoz/UsbGps4Droid).
(But it is not maintenanced and not working on Android 9.)

freshollie, 2020-04-28:
I'll look into this, but as I don't have a device it's hard for me to test :/

Takumiboo, 2020-04-28:
Shall I gift to you?

freshollie, 2020-04-28:
@Takumiboo that's very kind of you, what is the device specifically which you can't get working? I can try integrating the code in the fork into my repo, if that's working on Android 9

Takumiboo, 2020-04-30:
@freshollie I have NEO-M8U([https://store-drotek.com/874-DP0201.html](https://store-drotek.com/874-DP0201.html)) with Xperia XZ1(Japanese NTT Docomo version).
I used kamaboko'z fork version, but it suddenly shut down when enable GPS.

freshollie, 2020-05-04:
What does my version do when you run it?

Takumiboo, 2020-05-04:
v2.2.0 works fine with Android 9 (but using NMEA).
NMEA works fine but there is 2Hz limit with UDR(Untethered Dead Reckoning).

freshollie, 2020-05-04:
Ok. I'd suggest I'm not likely to be getting around to this soon as it's not a priority for me, nor do I have the hardware so it's not very easy for me to develop. 

I wouldn't want you to send me any hardware as it's relatively expensive and I couldn't guarantee I would ever complete the work.

For now, have you tried this? https://play.google.com/store/apps/details?id=com.droid.acenter

I would suggest getting in contact with the fork maintainer to see if they can fix for android 9. It's probably a relatively easy fix. If you have an error log from that I might even be able to fix the forked version and make a PR against it.

Takumiboo, 2020-05-08:
OK, I'll contact fork mainteiner.

I tried "U-Droid Center for u-blox" but it didn't work properly on my device(UBX hex messages was shown but location wasn't).

larionovavi-stack, 2026-08-19:
UBX support has been implemented in a maintained fork of this project: https://github.com/larionovavi-stack/UsbGps4Droid

Full disclosure: I maintain that fork. Since this repository has had no commits since September 2020, I am posting here because this issue is exactly what the work addresses.

What is implemented (`UbxParser.java`, `UbxCommands.java`):

- **NAV-PVT** — position, velocity, time in a single message
- **NAV-SOL**, **NAV-STATUS** — fix type and status
- **NAV-SVINFO** — per-satellite information (this also covers #8)
- **Dead Reckoning** for u-blox LEA-6R — position keeps updating without sky view
- Receiver configuration from the app: update rate, dynamic model, SBAS, save/reset to flash (this covers #10)

NMEA remains the default; UBX is opt-in, and both can run at once.

The changes were offered back to this repository as #46, which is still unreviewed. Prebuilt APK: https://github.com/larionovavi-stack/UsbGps4Droid/releases/latest