freshollie/UsbGps4Droid #22: Deactivate confirmation dialog for Android USB host authorization

dslthomas, 2020-07-21:
Sorry for my bad english but I have to use the google translator.

I have a rooted Galaxy S9 with Android 10 and am trying to get a Navilock NL-442U USB 2.0 GNSS GPS receiver SiRFstarIV to run. I looked at the post of the proposed solution, but it doesn't seem to work with Android 10.

Can you tell me how I can fix the problem? Unfortunately I am not a developer.

Thanks and greetings from Hamburg

[image]

[image]

[image]

[image]

dslthomas, 2020-07-22:
Is there really no solution here? I've been trying all sorts of tips from the Internet for 2 days now but none really works. Everything doesn't seem to work for Android 10.

freshollie, 2020-07-23:
Hey @dslthomas, I'll have a look at android 10 and USB issues, but I am confused by what your exact problem is. Can you describe what happens when you select your USB device and click start? Is that the message that pops up? Does the app crash? Are there any messages in the app? Does the app run but you have no location updates?

Ich spreche nur ein bischen Deutsch, so mussen wir Englisch sprechen :(

dslthomas, 2020-07-24:
Hello, thanks for your help!!!!

I use the Galaxy S9 in the car on a docking station and an external display. The Navilock NL-442U USB 2.0 GNSS GPS receiver SiRFstarIV receiver is connected to the docking station.

https://www.youtube.com/watch?v=w6l-g9UskWk

https://www.youtube.com/watch?v=vN-myYNgF-c

But I have to confirm the message in the APP with OK. Otherwise the GPS will not work:

https://www.youtube.com/watch?v=nrQpaVCEZvU

The APP itself runs absolutely reliably. The app doesn't crash either. My problem is the USB problem. I have to confirm the message with OK every time.

I have full root rights and a LineageOS ROM installed on the Galaxy S9.

freshollie, 2020-07-24:
Have you tried doing this? https://stackoverflow.com/a/30563253/1741602

I have no idea if this works for Android 10, but it works for older versions of ASOP android. The aim is to modify your system UI so that it always accepts the USB permissions before you see the popup.

dslthomas, 2020-07-24:
> Hast du das versucht? https://stackoverflow.com/a/30563253/1741602

Yes, I tried that. Unfortunately I have no idea what I'm doing. I did not find the relevant lines and then rejected them. I'm just not a programmer.

freshollie, 2020-07-26:
Can you send the smali code over gist.github.com? I'll take a look to see if I can find anything to disable it.

dslthomas, 2020-07-26:
I hope I've done it right now:
[https://gist.github.com/dslthomas/6847b790c4023660bccf6a245fc01ffb](url)

freshollie, 2020-07-26:
Looks like the relevant line was changed from `invoke-virtual {p0}, Lcom/android/systemui/usb/UsbPermissionActivity;->setupAlert()V` to `invoke-virtual {p0}, Lcom/android/internal/app/AlertActivity;->setupAlert()V`: 

https://gist.github.com/dslthomas/6847b790c4023660bccf6a245fc01ffb#file-usbpermissionactivity-smali-L414

Look at the rest of the code, you should be able to follow the same instructions. Please take a backup of your old jar before attempting this.

dslthomas, 2020-07-26:
OK, thanks for your help, but that's too high for me. I have no idea what I'm doing here. I'm just not a developer. Copy a few files back and forth and a few Linux commands work, but that's far too high for me here. The whole computer just crashed when I tried to pack this classes.dex again. Then unfortunately I have to look for another solution.

[image]

dslthomas, 2020-07-26:
For illustration only: almost all tabs come from the attempted solution. This is what my browser looked like for the last 5 days. Now I have to surrender, unfortunately.

[image]

freshollie, 2020-07-26:
Developing headunits is very hard. I know the struggle. Good luck with your solution anyway! To do this project, you will learn lots about programming! It's unfortunate that android requires these popups.

dslthomas, 2020-07-27:
So, short feedback - the problem is solved. I looked at it again today with a colleague. If you currently download Smali and Baksmali like me, the following commands must be used:

**java -jar baksmali.jar disassemble c:\adb\classes.dex
java -jar smali.jar assemble out -o c:\adb\classes.dex out**

then it works too. I then only replaced the classes.dex via 7Zip and replaced the SystemUI.apk with the original ones. Now it works without a query!

1000 thanks!

freshollie, 2020-07-27:
Brilliant! Glad your problem is finally solved!

closed 2020-07-27 by freshollie