freshollie/UsbGps4Droid #23: Wrong GPS time

slavka93, 2020-09-11:
Hello. I live in Russia, Moscow region. I want to express my deep gratitude for your application. Everything works great. Only after turning on the GPS item, the time is set to +12 hours and the region +3 East Africa. You have the opportunity to edit this time adjustment in the application, otherwise in the navigation programs the themes of the night version are installed during the day. Thank.

slavka93, 2020-09-11:
UsbGps4Droid-v2.2.0-release.apk

freshollie, 2020-09-11:
Hi. Are you using the gps time to set the system time? Is that causing the issue?

slavka93, 2020-09-11:
Yes

freshollie, 2020-09-11:
Think I've found the issue, but I don't have the hardware with me to check. I will make the change, and compile an APK which you can test here.

slavka93, 2020-09-11:
Yes, of course, I will test everything and write about the results.

slavka93, 2020-09-12:
Hello. Now I noticed that the program took 12 hours from the actual time. It will take 12 hours, then it will add. It happens that the time does not change and it corresponds to the actual one.

freshollie, 2020-09-12:
Hey, sorry I am a bit confused. You might be able to explain better with screenshots :)

freshollie, 2020-09-12:
Also, can I confirm are you on a GMT+12 timezone?

slavka93, 2020-09-12:
I am in time zone +3. Photos will now be.

slavka93, 2020-09-12:
[image]
In this photo, the top line is the actual time. The bottom line is -12 o'clock.

slavka93, 2020-09-12:
[image]
In this photo, the top line is the actual time. The bottom line is +12 o'clock.

slavka93, 2020-09-12:
And the date is always wrong, but it's not important to me.

freshollie, 2020-09-12:
Can you try this version? [UsbGps4Droid-v2.2.0-release.apk.zip](https://github.com/freshollie/UsbGps4Droid/files/5212635/UsbGps4Droid-v2.2.0-release.apk.zip)

slavka93, 2020-09-12:
Yes now.

slavka93, 2020-09-12:
Thank!!!! It works !!!! Out of 5 system starts, 5 correct readings.

slavka93, 2020-09-12:
Please tell me. Is it possible to run the program in the background? Android 9.

slavka93, 2020-09-12:
By adjusting the time, I'll go for a ride, I'll see.

freshollie, 2020-09-12:
Not sure on your issue with the background, can you open another issue to track? I'm gonna close this as I've published v2.2.1 with the fix.

closed 2020-09-12 by freshollie

slavka93, 2020-09-13:
Everything works great! Many thanks!