freshollie/UsbGps4Droid #24: Only showing 12 satellites

invisible789, 2020-09-29:
I've attached gnss module which caters 3 satellite constellations gps+ glonass + galeleio, i checked with other app and i was connected to 26 satellites, whereas gpsusb shows 12 only, should it not show 26 when I'm connected to 26 satellites??

freshollie, 2020-09-29:
Hey @invisible789 can you send a screenshot of the log data which is in the app? Feel free to redact your coordinates, they should be obvious. Just need to understand what data is coming back from the module, as I just read the number of satellites from that.

invisible789, 2020-09-29:
sure, i'm at work, i'll surely share when i get back home by evening, Thanks for prompt response

invisible789, 2020-09-29:
Here's my nmea data
https://mega.nz/file/qsZVXQZT#UgrsY69vnpMXOjJGZ8rewr2pTSfrXM1w_q4fsFLkdcg

invisible789, 2020-09-30:
@freshollie 
Sir did you checked my nmea log data?

freshollie, 2020-10-01:
Apologies, not had a chance to check. Will take a look tonight.

freshollie, 2020-10-05:
@invisible789 as per NMEA docs
> It is reasonable for the GSV sentence to contain more satellites than GGA might indicate since GSV may include satellites that are not used as part of the solution.

The number of satellites displayed is the number of satellites actually used to calculate the position, not the number of satellites in view.

There is a feature branch for displaying this satellite information, where I use GSV so it can contain all the information about the satellites.

closed 2020-12-10 by invisible789