freshollie/UsbGps4Droid #26: Date error.

slavka93, 2020-12-01:
Hello. Every time after loading the device, there is an error in the date. It is permanent 2020.05.21 Is it possible to fix this error?

slavka93, 2020-12-01:
[image]
Actually the date is 2020.12.01.

freshollie, 2020-12-01:
Very interesting. Are you able to provide an NMEA log? If you enable in the settings then a file should be saved to your device storage?

slavka93, 2020-12-01:
I will do everything as you write. But where can I find it later in the device?

freshollie, 2020-12-01:
It should exist in the storage folder when you connect via USB

slavka93, 2020-12-03:
[dmesg.txt](https://github.com/freshollie/UsbGps4Droid/files/5636447/dmesg.txt)
Hello. Sorry, I couldn't before.

slavka93, 2020-12-06:
Hello. Need a new file?

freshollie, 2020-12-06:
Hey @slavka93, I've taken a look at the source (it's been a while since I've worked on this) and the date from the GPS device us currently not used for a fix, only the time. It is possible to use the date too, but I can't promise when I can add this,

slavka93, 2020-12-07:
Hello. Well I will wait. Thanks.

slavka93, 2021-04-07:
Hello. Please help me solve this problem. Thanks.

freshollie, 2021-04-07:
Apologies I completely forgot about this!

slavka93, 2021-04-08:
It's okay, I'm ready to wait.

freshollie, 2021-04-25:
This may be more complicated. Can you check if you receive a `RMC` or `ZDA` message types?

slavka93, 2021-04-25:
Hello. Need to check the boxes and uncheck the log file?

slavka93, 2021-04-26:
Hello. How to check? There is nothing in the characteristics.