freshollie/UsbGps4Droid #27: High refresh rate (5Hz) with baud 115200 causing shut down eventually

invisible789, 2020-12-10:
@freshollie 
I was thorouly enjoying this app with settings Baud 115200 at 2Hz and data read is always fine and very light to system, recently i upgraded to 115200 at 5Hz, initially, it looked fine but after continuous use for 15 minutes , it started to show nmea sentences at very stuckable speed display and eventually app closed itself further after 10 minutes,, again tried to reproduce the same and it happened again, i'm using CP2102 Sil Lab with Ublox Neo M8N, both look fine by specs for set speed and Baud, Can you please look into this issue, as i was looking forward for 10 Hz at Baud 230400, which is not an available option with the app as well. Just got to my mind if Nmea sentences flushing right or even an option(not available) to turn OFF Nmea sentences display might help? 

As we have Ublox devices for multiple Layer connectivity with Multiple Constellations, pushing Baud and Refresh rate requirments to 3M Baud, Hope to get your expert opinion and resolution of the issue 

Regards

freshollie, 2020-12-10:
Hmm, could you get a logcat when this occurs? It would be nice to see where abouts the issue is.

invisible789, 2020-12-11:
> Hmm, could you get a logcat when this occurs? It would be nice to see where abouts the issue is.

Here's my NMEA data, i was unable to produce issue with same intensity in which app closes itself, but slowness in logs is there, i have a few questions , please help me understand those things, first GGA shows it's locked to 12 satellites whereas other apps lock to 26/27 satellites , why so?

Secondly, what NMEA messages are needed for app to work properly, you can find complete lists logging in data attached, GGA, GSV, VTG, GST etc, please guide which messages i can turn off safely? Without compromising anything

freshollie, 2020-12-12:
> please help me understand those things, first GGA shows it's locked to 12 satellites whereas other apps lock to 26/27 satellites , why so? 
Please see my explanation of this here https://github.com/freshollie/UsbGps4Droid/issues/24
And http://aprs.gids.nl/nmea/#gga for more information

I'll get back to you on the required sentences, I'm not exactly sure what is causing the slowness issue, a logcat would still be helpful if you can get me that.

invisible789, 2020-12-15:
> Please see my explanation of this here #24
> And http://aprs.gids.nl/nmea/#gga for more information
>
> I'll get back to you on the required sentences, I'm not exactly sure what is causing the slowness [...]

did you check required sentences as i have activated most of them, including VTG GLL GST GLL GSA etc, i want to just focus on required sentences and disable those unrequired, any info please?

freshollie, 2020-12-15:
Looking in here https://github.com/freshollie/UsbGps4Droid/blob/master/app/src/main/java/org/broeuschmeul/android/gps/nmea/util/NmeaParser.java

RMC and GGA seemed to be the only requirements for a GPS fix. Though the issues you are experiencing, I would suggest there is a deeper root cause than the number of sentences being sent

eharvester, 2020-12-18:
I just installed it, have it at 115kBaud /10Hz running, it has been stable for hours so far. I am using Android 7.1.1 on my Samsung SM-T377, PL2303HX and REYAX RY8253F. No problems, plug'n'play.
Great app, keep up the good work!

freshollie, 2020-12-30:
@invisible789 are you able to get me a system log? I would like to see the errors which may have caused the application to crash for you

invisible789, 2020-12-30:
@freshollie
I'll grab the system log for you after new year, I'm closing this issue right now and come back with system log later on if needed, Thanks for being super supportive :) HAPPY NEW YEAR EVERYONE

closed 2020-12-30 by invisible789