freshollie/UsbGps4Droid #28: A display issue

franklinbill, 2020-12-28:
Dear developer!
In the process of using, I found a bug in layout / display. As shown in the figure, in the first page, the text in the red box is half blocked. This may be an adaptation problem. I suspect it is because **display and font size** are set to large (mainly due to **display size**).
Because of some problems with my eyesight, I often use the largest font and display.
[image]
App version is 2.2.1, my mobile phone is Google nexus 6, screen resolution setting is 1920* 1080, **_font size and display size are the largest ! ! !_**, Android system is 8.0.
The above is a description of the problem. The overall experience of the software is good. You can see that the design is very attentive. Thank you very much for the efforts of the developers all the time. I hope to receive a reply.
Thank you!
[image]

freshollie, 2020-12-30:
Just to confirm, are you actually using this application? I understand this may be a bug, but the specific use case for this application isn't for accessibility purposes, and I can see you've opened other issues in lots of other repos with exactly the same bug? What's the motivation here?

franklinbill, 2020-12-30:
Thanks*100000 for your reply! Indeed, this kind of display problem is also widespread in other apps. The main reason why I pay attention to these bugs is that with the Aging of the population, more and more elderly people begin to use these apps. For example, my parents often use the largest display size and font size due to the deterioration of their eyesight, and they often encounter these display bugs. Although these problems will not cause crash, they will reduce the user experience of the elderly. At the same time, these bugs are easy to fix, so I hope developers can fix these bugs to improve the user experience of the app. I also think that these bugs will be taken seriously in the future.

freshollie, 2020-12-30:
That makes sense, but I think you misjudge the audience which would be using this application, this isn't for the general population. 

> The overall experience of the software is good. You can see that the design is very attentive

I question the integrity of the post, as using this application requires specific hardware which you don't seem to poses, as such I don't think complimenting something which you've not had a chance to use is very honest ;). 

My concern here is it seems you are pretty familiar with the problem, and thus your efforts might be better placed at submitting PRs to repositories to fix the problem instead of submitting the same template issue to hundreds of repos. If you know the issue, as you say it's not so complex to fix, then a PR would provide value. 

A screenshot of a bug in an application which you don't actually use doesn't create value, it wastes (most) peoples time.

franklinbill, 2020-12-30:
Thank you very much for your reply and suggestions, I will also consider them, but I don't think this work is meaningless(it wastes (most) peoples time). Of course, you can have different opinions. I just hope we can pay attention to these problems, even if it doesn't seem so "important" now.