freshollie/UsbGps4Droid #30: No data from USB receiver to GPSTEST an WAZE

fejohn1231, 2021-02-14:
Dear Developer.

I have Rockchip PX5 car head unit with Android 9. The internal AT6558 GNSS version receives only the GPS and chianian satellities.
I have decided to applicate an VK-168 USB external reciver with your apk. Connecting the USB the receiver will be ttyACM0 and go well, I see the NMEA sentences. But into the android services no arrives anythig. The internal receiver is ttyS0 -> ttyGps.
Can yuo resolve this problem?

Thanks
Janos

freshollie, 2021-02-14:
Hey! Can I have some more details here? When you select the device to start the service what happens? I can I get a screenshot? Does the app show the device as "connected"? Does it show an error?

fejohn1231, 2021-02-14:
Hi!With the internal gps operates normally. On ttyGps and ttyS0 come the NMEA sentences. On Gpstest I see 10 satellites, fixed are 8-9.When I start UsbGps4Droid choosing the USB receiver, Gpstest and Waze say no Gps. Looking ttyACM0, data arrive, in window of UsbGps4Droid seem data coming from USB receiver.On ttyS0 and ttyGps seems data coming from internal chip.Gpstest no satellites in view and no fixed.
I think, the your apk closes the connection between internal chip and services, but not connetcs the USB to the service. 

Yes, your app says device is connected, I can choose regurarly.
No error message, only the Gpstest or waze are without data.

I can send screen foto tomarrow, because the HU in in my car.

freshollie, 2021-02-14:
Do you see sentences in the application main screen when you are connected?

fejohn1231, 2021-02-14:
Yes, I see in your app's main screen

freshollie, 2021-02-14:
That's very weird. Are you sure that you have the mock locations settings setup?

fejohn1231, 2021-02-14:
Yes, I choosed your apk as mock location application

freshollie, 2021-02-14:
Hmm, that's very weird. And you can see your location lock in the app?

fejohn1231, 2021-02-15:
Hi,

I made foto:

This is the dev folder, where ttyACM0 is the USB receiver, the ttyS0 is the internal receiver (AT6558 chip)
[image]

These are the sentences from the USB receiver.
[image]

These are from internal receiver.
[image]

Your application with choosed receiver
[image]

Your started app
[image]

Gpstest with started UsbGps4Droid:
[image]

UsbGps4Droid switch off:
[image]

GpsTest with stopped UsbGps4Droid
[image]

freshollie, 2021-02-15:
Very weird. What I would say is that GPSTest is never going to pick up satellites because that information can't be published to android. Do any applications with GPS usage pick up the location?

fejohn1231, 2021-02-15:
I tried several GPS testing apps, nothing used the data. When the application starts, ttyGps -> ttyS0 in the dev folder disappears. So I thought I'd redirect ttyACM0 to them.

freshollie, 2021-02-16:
Have you tried another mock location app? It could be that your android OS doesn't support mock locations properly :/