freshollie/UsbGps4Droid #32: system time adjustment

pepelillo, 2021-04-28:
Hello. 
I have issues with adjusting system time from GPS time, I'm in android 9 x86-64. Everything works but this, (and the ugly usb permission popup every boot).
I've download 2.2.1version hoping that the issue is solved, but unfortunately seems not to be 2.2.1, is 2.2.0, I've downloaded several times but there is a missmatch version.
Thanks

freshollie, 2021-04-28:
Hey! The USB permission popup is not solved with `2.2.1`, what exactly is the issue?

pepelillo, 2021-04-28:
I know, maybe I'm not clear, my issue is with time sync (adjust system time with GPS time) it does not work in my case, Yes my device is rooted and root permission is granted to your app. The USB permission is more a wish.....:-) I'm trying to automate the process with Macrodroid and ui interaction , but is not so easy because the timing of the popup, delay does not seem to work.

pepelillo, 2021-04-28:
And the last release is not maked as 2.2.1 but 2.2.0 in app settings

freshollie, 2021-05-05:
> it does not work in my case

So it sets the time on `2.2.0` but not on `2.2.1`?

pepelillo, 2021-05-10:
No, actually I never get system time and date sync with GPS date and Time, I use 2.2.0, and when I download 2.2.1 realize that in fact it was 2.2.0 (at least this is the version reported by Android), And then no improvement. Can you check that the version uploaded is 2.2.1?

freshollie, 2021-05-14:
In that log that you see in the main app screen, can you see the `GPRMC` message type? Can you send me a screenshot of the logs?

`2.2.1` didn't make a significant change to fix this. It only made it work correctly with 24 hour time.