freshollie/UsbGps4Droid #33: device open not working

ce-4, 2021-05-08:
I have an older MTK based usb/bt receiver (outputs nmea) that uses the CDC-ACM driver. When I open it via this app, nothing happens.

However, when I first open the device once via https://play.google.com/store/apps/details?id=de.kai_morich.serial_usb_terminal (which shows nmea messages coming in correctly) and then close usb-terminal and reopen it in usbgps *then* all works fine and gps location is made available as expected. Once I disconnect the usb and reconnect I have to redo the procedure. Any hints what could be missing in the initialization phase?

freshollie, 2021-05-14:
Hey! I've had a look, and this app would probably need a large amount of refactoring in order fix this.

The app you reference uses [this library](https://github.com/mik3y/usb-serial-for-android), which probably provides support for more boards. I've used this library before, but it would probably be a large amount of work to get it working.

I currently don't have a device which I can use to develop this, or the free time to make such a large change :(

ce-4, 2021-05-23:
No worries, i just resolved my issue and bought a supported ublox usb receiver for cheap. Works fine woth the new one.

HighPriest, 2025-04-11:
@ce-4 
> When I first open the device once via https://play.google.com/store/apps/details?id=de.kai_morich.serial_usb_terminal (which shows nmea messages coming in correctly) and then close serial-usb-terminal app and reopen it in usbgps *then* all works fine and gps location is made available as expected.

Thank you kind soul for this comment!  It is alive!
No application worked with my Quectel GPS module, not the GPS Connector, not Usb2GPS. But, starting the Serial Terminal, setting correct Baud Rate in that application & then handing over the connection to USB2GPS, works flawlessly. Many thanks 🤝