freshollie/UsbGps4Droid #34: gps week number rollover fix

NH-Networks, 2021-06-03:
can you implement a gps week number rollover fix ??

freshollie, 2021-06-03:
I don't know what this means, can you provide more detail?

NH-Networks, 2021-06-03:
https://forum.xda-developers.com/t/joying-intel-sofia-head-unit-is-pulling-a-2001-date-from-gps-and-does-not-obey-network-time.4286581/#post-85124649

i'm hoping your tools can fix this... and will work with the gps...

freshollie, 2021-06-03:
If you have root access, this app can set the system time to the GPS time. Not 100% sure this will stop the system from still using the inbuilt GPS time though.

NH-Networks, 2021-06-03:
> If you have root access, this app can set the system time to the GPS time. Not 100% sure this will stop the system from still using the inbuilt GPS time though.

i have root access, the problem is dat the time from gps is incorrect due to that bug and 1024 weeks need to be added, see jave code example

freshollie, 2021-06-03:
I saw, but is this specific app causing this? If not, I think this app will not have the same issues as described

NH-Networks, 2021-06-03:
No it's not app related, that app needs a fix also in code but i have no idea how... changes are needed in decompiled smali files i guess... 

But time sync can be turned off in settings so that app will no overrule it... now i only need a other app that runs as a service that prevent time and date to fall back to  "today date - 1024 weeks" 

for more info on the issue see here: 

https://www.reddit.com/r/LineageOS/comments/edz2oa/gps_week_number_rollover_bug_can_it_be_solved_in/

freshollie, 2021-06-03:
Ah, tbh I've just had a look at my code and I don't even parse the "date" only the UTC time, so I don't think there is anything I can fix here.

NH-Networks, 2021-06-03:
> Ah, tbh I've just had a look at my code and I don't even parse the "date" only the UTC time, so I don't think there is anything I can fix here.

ah dammit, had my hopes up that you tool could help me out, if you don't mind me asking is there anything you can do to help me out?