freshollie/UsbGps4Droid #37: GPS switched off by default

robbo69, 2022-07-23:
Currently, when app starts up the GPS on/off toggle is set to OFF. Can there be an option so that at startup the toggle is at ON? At present I have this app autostart but I still have to go to it to start up GPS - it would be nice to have this happen automatically in the background.

freshollie, 2022-07-23:
Hey!

If you are auto starting the app, have you considered using Tasker to send the intent to start the app in the background instead? 

https://github.com/freshollie/UsbGps4Droid#service-behaviour

I could dedicate some time to adding this function but background start would likely be better for your usecase

robbo69, 2022-07-24:
Thanks for the fast reply.  I already start the app in the background using the Autostart app. What I want is not to have to go to UsbGps4Droid to switch on GPS but for the app to do that in the background by having the on/off GPS toggle set to on when the app starts up.

freshollie, 2022-07-24:
> I already start the app in the background using the Autostart app

Sure, but you should use an app instead which allows you to send that intent.

I can add the functionality to start on app open, but I'm currently not focused on this app so I might not get around to this for a while. 

Can you try using Tasker to send an intent? There is some docs here https://tasker.joaoapps.com/userguide/en/intents.html

robbo69, 2022-07-25:
I'll give Tasker a try but I'm not sure how it will go, as these Chinese head units hide android behind their MCU and often cripple basic functions (eg bluetooth), so I would be grateful if you could modify your app when you get time, as it would be a simpler option than adding Tasker (if it works) to start on boot.