freshollie/UsbGps4Droid #43: Could NOT work on HUAWEI matepad (HarmonyOS)

bilibili12433014, 2024-06-13:
Device: matepad 11.5S 8G+256G
OS Version: HarmonyOS 4.2.0.126

Detail:
SoftWare(UsbGps) successfully reads GPS data from my U-blox, but my navigation softwares could NOT read GPS information.

freshollie, 2024-06-13:
Did you do this? https://developer.huawei.com/consumer/en/doc/HMSCore-Guides/location-develop-steps-0000001050746143#section5825644497

bilibili12433014, 2024-06-13:
I do have set mock app, but it did not work!

bilibili12433014, 2024-06-13:
I found difference between your code and the code in the link. Is this a possible cause of this bug?
your code:
```
<uses-permission
        android:name="android.permission.ACCESS_MOCK_LOCATION"
        tools:ignore="MockLocation" />
```
code in the link:
```
<uses-permission
    android:name="android.permission.ACCESS_MOCK_LOCATION"
    tools:ignore="MockLocation,ProtectedPermissions" />
```

freshollie, 2024-06-13:
Yes it could be. Are you able to recompile the app yourself and test?

bilibili12433014, 2024-06-13:
let me have a try

bilibili12433014, 2024-06-13:
This may be the problem:
[image]

After I added `,ProtectedPermissions` , no error appeared. But I could not compile, could you help build a release?

freshollie, 2024-06-13:
I think this repo is on API level 23 so should be ok. Why can't you compile? I've not got the repo set up on any of my current machines

freshollie, 2024-06-13:
Btw, if you are able to select the app in mock location this is not likely to be the issue. I'm not sure if there is a specific reason for mock location not working on that OS, it would require some local debugging from you

bilibili12433014, 2024-06-14:
Now, I successfully sync your app and compile it. However, I could not install app to my matepad, because Android Studio could not recognize it. As Android Studio could recognize my phones, install app and run app (one is also HarmonyOS but 4.0.0, one is Xiaomi HyperOS), i think, this may be a HarmonyOS bug about USB DEBUG and may affect GPS MOCK. I am trying to communicate with HUAWEI's engineer, if I get any futher information, I will make a new comment.

bilibili12433014, 2024-06-14:
By searching Internet, there is a bug about USB DEBUG: "file" mode must be selected, if "charge" mode is selected, USB DEBUG will not work; but I still could not find the reason why GPS MOCK could not work.

freshollie, 2024-06-14:
I'm not sure. Have you tired another app with GPS mock? Is the GPS location being displayed in the app?

bilibili12433014, 2024-06-22:
After test, I have confirmed that your application runs correctly and there are no bugs. The reason the mock location is not working is that my navigation software obtains location data directly from the satellite, not from the operating system. I wrote a demo to obtain location data and tested it, and the mock location feature does indeed work. Other mock applications and navigation softwares are tested, they do work.

closed 2024-06-22 by bilibili12433014

freshollie, 2024-06-25:
Ok, thanks for letting me know :)